micro-nova/AmpliPi #981: Source state isn't being set properly during stream close / preset invocation

SteveMicroNova, 2024-10-14:
In #980 and #979 I found that closing a stream doesn't always set that source.input to a valid option, figure out why state isn't being set properly in different cases involving presets and stream closures

rtertiaer, 2024-10-15:
I'm not sure there's enough information here for someone other than yourself to solve this issue; can you elaborate on the problem as you've witnessed it and perhaps create a repro case?

SteveMicroNova, 2024-10-15:
> I'm not sure there's enough information here for someone other than yourself to solve this issue; can you elaborate on the problem as you've witnessed it and perhaps create a repro case?

For some reason [this](https://github.com/micro-nova/AmpliPi/pull/979#discussion_r1800019407) solved the issue I was having, and illuminated this one. When streams are closed manually or via an empty preset on streamer, sometimes they get set to an invalid state that allows them to path through the "unknown" path of this function:
```
export const getSourceInputType = (source) => {
    // can return:
    // none, rca, digital, unknown
    if (!source || !source.input || source.input == undefined) return "unknown";
    const input = source.input.toLowerCase();

    if (input === "none" || input === "" || input === "local") {
        return "none";
    }

    const split = input.split("=");
    if (split.length !== 2) {
        console.log(`WARNING: unknown source input: ${input}`);
        return "unknown";
    }

    if (split[1] < 1000) {
        return "rca";
    } else {
        return "digital";
    }
};
```

I didn't get very far into it myself, to witness this you'll just have to keep an eye on the state of the various sources while closing stuff on the streamer at minimum, I haven't been able to look at a main unit yet