micro-nova/AmpliPi #1016: Zone index is wrong when saving settings

121940kz, 2025-01-10:
in Zones.jsx

`setAlertText(`Zone ${zone.id} updated successfully!`);`

and 

`setAlertText(`Zone ${zone.id} update failed...`);`

These need updated to be the zone.id + 1  because when you save the zone, they are 0 based indexed so the message at the bottom of the app is incorrect.

linknum23, 2025-01-10:
Nice catch! This printout really doesn't make sense with the initial names that zones are given. A more useful message probably looks something like:

**Zone 1 (id=0) updated successfully!**

Which in general this would look something like:

**ZONE_NAME (id=ID) updated successfully!**

SteveMicroNova, 2025-01-10:
Good catch!
I've made a solution to this in #1017, which should make it out in the next update.

closed 2025-01-10 by SteveMicroNova