micro-nova/AmpliPi #1019: Amplipi bootloop following restart

wgordon17, 2025-01-11:
Attempted a 0.4.5 update, however that failed. Assuming this is similar to #1005, it was something about errors copying.

So, I figured (like all techie's)...that's something that a **reboot** should solve! 💪

Upon the system coming back up, it now looks like Amplipi is in a bootloop looking for `celery`

```
Jan 10 19:51:30 amplipi systemd[578]: Started Amplipi Home Audio System.
Jan 10 19:51:35 amplipi authbind[3874]: Traceback (most recent call last):
Jan 10 19:51:35 amplipi authbind[3874]:   File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
Jan 10 19:51:35 amplipi authbind[3874]:     "__main__", mod_spec)
Jan 10 19:51:35 amplipi authbind[3874]:   File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
Jan 10 19:51:35 amplipi authbind[3874]:     exec(code, run_globals)
Jan 10 19:51:35 amplipi authbind[3874]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/uvicorn/__main__.py", line 4, in <module>
Jan 10 19:51:35 amplipi authbind[3874]:     uvicorn.main()
Jan 10 19:51:35 amplipi authbind[3874]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/click/core.py", line 1137, in __call__
Jan 10 19:51:35 amplipi authbind[3874]:     return self.main(*args, **kwargs)
Jan 10 19:51:35 amplipi authbind[3874]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/click/core.py", line 1062, in main
Jan 10 19:51:35 amplipi authbind[3874]:     rv = self.invoke(ctx)
Jan 10 19:51:35 amplipi authbind[3874]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/click/core.py", line 1404, in invoke
Jan 10 19:51:35 amplipi authbind[3874]:     return ctx.invoke(self.callback, **ctx.params)
Jan 10 19:51:35 amplipi authbind[3874]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/click/core.py", line 763, in invoke
Jan 10 19:51:35 amplipi authbind[3874]:     return __callback(*args, **kwargs)
Jan 10 19:51:35 amplipi authbind[3874]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/uvicorn/main.py", line 449, in main
Jan 10 19:51:35 amplipi authbind[3874]:     h11_max_incomplete_event_size=h11_max_incomplete_event_size,
Jan 10 19:51:35 amplipi authbind[3874]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/uvicorn/main.py", line 569, in run
Jan 10 19:51:35 amplipi authbind[3874]:     server.run()
Jan 10 19:51:35 amplipi authbind[3874]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/uvicorn/server.py", line 60, in run
Jan 10 19:51:35 amplipi authbind[3874]:     return asyncio.run(self.serve(sockets=sockets))
Jan 10 19:51:35 amplipi authbind[3874]:   File "/usr/lib/python3.7/asyncio/runners.py", line 43, in run
Jan 10 19:51:35 amplipi authbind[3874]:     return loop.run_until_complete(main)
Jan 10 19:51:35 amplipi authbind[3874]:   File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
Jan 10 19:51:35 amplipi authbind[3874]:     return future.result()
Jan 10 19:51:35 amplipi authbind[3874]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/uvicorn/server.py", line 67, in serve
Jan 10 19:51:35 amplipi authbind[3874]:     config.load()
Jan 10 19:51:35 amplipi authbind[3874]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/uvicorn/config.py", line 477, in load
Jan 10 19:51:35 amplipi authbind[3874]:     self.loaded_app = import_from_string(self.app)
Jan 10 19:51:35 amplipi authbind[3874]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/uvicorn/importer.py", line 24, in import_from_string
Jan 10 19:51:35 amplipi authbind[3874]:     raise exc from None
Jan 10 19:51:35 amplipi authbind[3874]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/uvicorn/importer.py", line 21, in import_from_string
Jan 10 19:51:35 amplipi authbind[3874]:     module = importlib.import_module(module_str)
Jan 10 19:51:35 amplipi authbind[3874]:   File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
Jan 10 19:51:35 amplipi authbind[3874]:     return _bootstrap._gcd_import(name[level:], package, level)
Jan 10 19:51:35 amplipi authbind[3874]:   File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
Jan 10 19:51:35 amplipi authbind[3874]:   File "<frozen importlib._bootstrap>", line 983, in _find_and_load
Jan 10 19:51:35 amplipi authbind[3874]:   File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
Jan 10 19:51:35 amplipi authbind[3874]:   File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
Jan 10 19:51:35 amplipi authbind[3874]:   File "<frozen importlib._bootstrap_external>", line 728, in exec_module
Jan 10 19:51:35 amplipi authbind[3874]:   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
Jan 10 19:51:35 amplipi authbind[3874]:   File "./amplipi/asgi.py", line 27, in <module>
Jan 10 19:51:35 amplipi authbind[3874]:     import amplipi.app
Jan 10 19:51:35 amplipi authbind[3874]:   File "./amplipi/app.py", line 63, in <module>
Jan 10 19:51:35 amplipi authbind[3874]:     from amplipi.ctrl import Api, ApiResponse, ApiCode  # we don't import ctrl here to avoid naming ambiguity with a ctrl variable
Jan 10 19:51:35 amplipi authbind[3874]:   File "./amplipi/ctrl.py", line 41, in <module>
Jan 10 19:51:35 amplipi authbind[3874]:     import amplipi.streams
Jan 10 19:51:35 amplipi authbind[3874]:   File "./amplipi/streams/__init__.py", line 34, in <module>
Jan 10 19:51:35 amplipi authbind[3874]:     from .spotify_connect import SpotifyConnect
Jan 10 19:51:35 amplipi authbind[3874]:   File "./amplipi/streams/spotify_connect.py", line 13, in <module>
Jan 10 19:51:35 amplipi authbind[3874]:     from .. import tasks
Jan 10 19:51:35 amplipi authbind[3874]:   File "./amplipi/tasks.py", line 2, in <module>
Jan 10 19:51:35 amplipi authbind[3874]:     from celery import Celery
Jan 10 19:51:35 amplipi authbind[3874]: ModuleNotFoundError: No module named 'celery'
Jan 10 19:51:35 amplipi systemd[578]: amplipi.service: Main process exited, code=exited, status=1/FAILURE
Jan 10 19:51:35 amplipi systemd[578]: amplipi.service: Failed with result 'exit-code'.
Jan 10 19:51:36 amplipi systemd[578]: amplipi.service: Service RestartSec=100ms expired, scheduling restart.
Jan 10 19:51:36 amplipi systemd[578]: amplipi.service: Scheduled restart job, restart counter is at 1769.
Jan 10 19:51:36 amplipi systemd[578]: Stopped Amplipi Home Audio System.
```

Any suggestions?? 🙏

linknum23, 2025-01-11:
Have you tried upgrading a second time? We've seen issues like this before when too many python dependencies are introduced that are fixed with a second update. 

Even after a failed update the updater is still running. You can try reaching it at http://amplipi.local:5001/update

wgordon17, 2025-01-11:
It seems to think it's up to date now! 🤔 I _wonder_ if it's possible that my reboot could have interrupted the update process. Do I dare attempt an "Older Release" update to retry the 0.4.5 update??

[image]

linknum23, 2025-01-11:
Yup! That's exactly how you need to retry the update now that you have restarted.

wgordon17, 2025-01-11:
_Beautiful_! That did the trick, and I have it back up and running! 🙌 Thank you for your help, and _awesome_ job on creating a very robust system that can be debugged and fixed even when the main UI/thread is broken 🎉

closed 2025-01-11 by wgordon17