micro-nova/AmpliPi #1020: Internet Radio stops playing when Internet fails over

TerribleTesla, 2025-01-15:
Not a super important issue, but figured it was worth mentioning.

When an Internet Radio stream is playing and the primary internet goes down (Cable Modem goes out, and router fails over to backup ISP), the zone playing Internet Radio still shows as playing (Has a Stop button instead of the Play button) but nothing is playing anymore.

The failover takes ~5 seconds and the Internet Stream never starts back up without manual intervention.

Stopping and Starting the stream fixes the issue.

Please let me know if this is worth investigating (And what data I can collect to help).

linknum23, 2025-01-16:
The most helpful information for us right now would be the log file
generated after the issue occurs. You can find the internet radio log file
at `/home/pi/.config/.config/amplipi/srcs/SRC/log`, where SRC corresponds
to the source number (0, 1, 2, or 3) it was playing on.

While our `streams/runvlc.py` script includes code to handle such failures,
those specific scenarios haven't been thoroughly tested yet.

TerribleTesla, 2025-01-16:
I forced a failover and didn't get anything additional in the log:

```
/home/pi/.config/amplipi/srcs/1# tail -F log
Starting VLC instance.
Stream still opening; waiting 10.0 more seconds...
Stream still opening; waiting 9.75 more seconds...
Stream has opened.
Current track: zc3466 - None
```

Stopping and starting the stream via the WebUI just results in a cleared log file and the same thing again once it starts playing.

linknum23, 2025-01-16:
Based on the code in streams/runvlc.py it looks like the underlying python vlc client is still reporting the player status as playing. Here is the relevant code: https://github.com/micro-nova/AmpliPi/blob/f9a577ad1dd54b9b3ce25b885fb5e1868b8a7ee1/streams/runvlc.py#L214