micro-nova/AmpliPi #1021: Volume changes lag with multiple child zones

SteveMicroNova, 2025-01-20:
I've found that in CardVolumeSlider.jsx there is a logic flow that divides the individual zones into discreet api calls, so if you have every zone tuned in to the same source and you use the big volume slider on the home or player page it will go through zones 1-6 one by one. This makes it so that if you change the volume a lot quickly, it functionally queues up a bunch of individual zone volume updates and this can cause the actual auditory volume to lag a few seconds behind the user input and this compounds with every expander they have hooked up to the same source as well

Either batch those requests more efficiently or debounce them a little harder so you don't create this queue in the first place (or a secret third thing I've yet to think of)

linknum23, 2025-01-20:
Batching those requests should be as simple as using the /api/zones PATCH
endpoint since all of the zones are being updated to the same value.

SteveMicroNova, 2025-01-20:
> Batching those requests should be as simple as using the /api/zones PATCH
> endpoint since all of the zones are being updated to the same value.

They are not being set to the same value

The way that the parent volume bar works is by moving the child bars along the same delta, so if you move it from 50 to 60, it just adds 10 to all child volume bars unless it would make them go over 100. The group volume bars do authoritatively set volume rather than having this relativistic approach

linknum23, 2025-01-21:
We may want to add an optional field to a zone update such as
`vol_delta_f`, that pushes this behavior into the api.

closed 2025-02-20 by SteveMicroNova