micro-nova/AmpliPi #1037: Re-allow Channel Surfing of FM stations

vmooney, 2025-03-07:
I've got multiple radio stations streams defined and find it very inconvenient to have to stop the entire player to change channels.
Before [PR#901](https://github.com/micro-nova/AmpliPi/pull/901), it was possible to channel surf directly picking a different stream in the given player.

jevogel, 2025-03-10:
Hi @vmooney, have you tried this with the just-released `0.4.6`? I just checked and was unable to reproduce.

vmooney, 2025-03-10:
I confirm I am running 0.4.6 and tested it in an incognito window to avoid any issues with caching.

[image]

[image]

jevogel, 2025-03-10:
Ah, my mistake, I was looking at internet radio streams rather than FM hardware.