micro-nova/AmpliPi #1046: Group Volume Bars not calculating correctly

SteveMicroNova, 2025-04-28:
https://github.com/user-attachments/assets/d51c124b-7406-47cf-9919-e151c37d1eac

Group volume bars seem to be calculating their position based on the midpoint between the min and max, not the average of all zones

linknum23, 2025-04-29:
Yup, the API  reports the median of a group as the volume. It was designed this way.. The intention was for zone volumes to be able to increase past 100% so that a sequence of a large group adjustment increase than large decrease was not destructive. IIRC, reporting the median allows for the volume to be less effected when zones have a volume of over 100%. We have not added any support for increases past 100% that get saturated to t00% at a lower level yet. I just wanted to give you the reasoning for the current state.

chromics, 2025-05-21:
I want to contribute, but I can't seem to find the endpoint to run? This was all I managed to get...

[image]