micro-nova/AmpliPi #1054: Spotify-side volume changes are not realized

SteveMicroNova, 2025-09-02:
Previously when we were running [spotifyd](https://github.com/Spotifyd/spotifyd), Spotify volume controls functioned
spotifyd decayed over time so we swapped to [go-librespot](https://github.com/devgianlu/go-librespot), which does support volume changes but they sound notably worse than the same implementation in spotifyd so we made the decision to deactivate Spotify side volume support for now

This is while we await go-librespot's development so that they reach a point where we can have _volume propagation_, a feature that will allow us to merge the volume of Spotify and AmpliPi on those streams, making it so that they move as a single unit rather than requiring users to use both the Spotify and amplipi volume sliders to dial in their volume properly.

For devs, this issue is:
- Keep an eye on go-librespot's development
- add volume propagation functionality with go-librespot once available

vmachacek, 2025-09-03:
that's not good, maybe you should de-list support for SpotifyConnect (or add Asterix at least in the meantime)

burnsie-la, 2025-10-03:
> Keep an eye on go-librespot's development

Is there an issue for this in go-librespot's repo that I can watch?

SteveMicroNova, 2025-10-07:
> Is there an issue for this in go-librespot's repo that I can watch?

Not anymore, this is now entirely our problem. We just need to plug in the exposed functions and hopefully get this out in the next update or two